paulcarroty/indicator-upower #4: error reading /system/build.prop

Danfro, 2024-12-21:
The apps log shows this error here:

```
[21.12.24 22:16] /system/build.prop exists? True
[21.12.24 22:16] Traceback (most recent call last):
[21.12.24 22:16]   File "/opt/click.ubuntu.com/indicator.upower.ernesst.fork/current/indicator/ernesst-indicator-upower.py", line 440, in <module>
[21.12.24 22:16]     wi = UpowerIndicator(bus)
[21.12.24 22:16]   File "/opt/click.ubuntu.com/indicator.upower.ernesst.fork/current/indicator/ernesst-indicator-upower.py", line 86, in init
[21.12.24 22:16]     self.get_config_device()
[21.12.24 22:16]   File "/opt/click.ubuntu.com/indicator.upower.ernesst.fork/current/indicator/ernesst-indicator-upower.py", line 144, in get_config_device
[21.12.24 22:16]     build_prop_file = open("/system/build.prop")
[21.12.24 22:16] PermissionError: [Errno 13] Permission denied: '/system/build.prop'
```

OP5T, focal, devel.

paulcarroty, 2024-12-22:
Attach your `build.prop`.

Danfro, 2024-12-22:
[build.prop.txt](https://github.com/user-attachments/files/18221250/build.prop.txt)

I had to rename it to build.prop.txt to be able to upload here.

Danfro, 2024-12-22:
I should maybe add (not sure if that makes a difference), that I am running the custom kernel described here: https://forums.ubports.com/topic/9864/call-for-testing-a-the-new-build-variant-standalone-kernel-method-vs-full-system-image

Current plans are to make that default for OP5/OP5T with noble.

paulcarroty, 2024-12-22:
No idea how to identify your device, maybe something else in /system?

Danfro, 2024-12-22:
I did search around a bit. There is another file called build.prop in `/system/vendor/build.prop`. That actually holds those values:
```
ro.product.vendor.model=5T
ro.product.vendor.brand=OnePlus
ro.product.vendor.name=dumpling
ro.product.vendor.device=dumpling
```

paulcarroty, 2024-12-23:
Open the bugreport in your kernel repo.

closed 2024-12-23 by paulcarroty

Danfro, 2024-12-23:
Just wanted to let you know, that for some reason I can't set read permission on the build.prop files on my device. Not even via terminal. Sadly this means I am still running into the permission error and can't get the app working.

But I guess this is nothing the app needs to sort, but needs to be done on the port.

Thanks for your help and quick work to get my device supported.

paulcarroty, 2024-12-24:
`grep -rl "dumpling" /system/` ?

Danfro, 2024-12-24:
Something is not right on my end I guess...

--

Danfro, 2024-12-24:
Ah, the attached image didn't make it when replying via email. here it is

[image]

paulcarroty, 2024-12-24:
Weirdddd, how about `grep -rl "dumpling" /etc/`?

Danfro, 2024-12-25:
This returns the following output:
```
grep -rl "dumpling" /etc/
grep: /etc/sudoers: Permission denied
grep: /etc/.pwd.lock: Permission denied
grep: /etc/polkit-1/localauthority: Permission denied
grep: /etc/sudoers.d/README: Permission denied
grep: /etc/dhcp/ddns-keys: Permission denied
grep: /etc/gshadow-: Permission denied
grep: /etc/shadow-: Permission denied
/etc/system-image/channel.ini
/etc/deviceinfo/devices/halium.yaml
grep: /etc/gshadow: Permission denied
grep: /etc/security/opasswd: Permission denied
grep: /etc/ssl/private: Permission denied
grep: /etc/ppp/pap-secrets: Permission denied
grep: /etc/ppp/chap-secrets: Permission denied
grep: /etc/ufw/user6.rules: Permission denied
grep: /etc/ufw/before.init: Permission denied
grep: /etc/ufw/after.init: Permission denied
grep: /etc/ufw/before6.rules: Permission denied
grep: /etc/ufw/user.rules: Permission denied
grep: /etc/ufw/after.rules: Permission denied
grep: /etc/ufw/after6.rules: Permission denied
grep: /etc/ufw/before.rules: Permission denied
grep: /etc/shadow: Permission denied
```

I just removed some lines from networkmanager, also with "Permission denied".

And I compared to my suzu device, where uPower works fine, this commands output is quite similar there. My suzu hasn't got a build.prop in /system/vendor, but the one in /system has more read permissions than then one of my OP5T.

This seems to be more a porting issue as in which is correct and maybe expected by other services.

paulcarroty, 2024-12-25:
Attach both:
`/etc/system-image/channel.ini
/etc/deviceinfo/devices/halium.yaml`

Also let's check the permissions too: `ls -l /system/vendor/build.prop /system/build.prop`

Danfro, 2024-12-25:
The permissions for the two files don't allow reading I think:
```
-rw-------. 1 root root 2427 Nov  8 21:39 /system/build.prop
-rw-------. 1 root root 7237 Mar 23  2024 /system/vendor/build.prop
```
But I can't change them. I already tried that via terminal.

[channel.ini.txt](https://github.com/user-attachments/files/18247273/channel.ini.txt)

[halium.yaml.txt](https://github.com/user-attachments/files/18247274/halium.yaml.txt)

paulcarroty, 2024-12-27:
https://github.com/paulcarroty/indicator-upower/commit/08167ce83d4e10d0e16bad29593ff620f68e5008 should fix it.

Danfro, 2024-12-27:
Thanks a lot. Yes, now the indicator is working. :+1:

commit 38599d69 (2024-12-23): add OP5T workaround; close #4