paulcarroty/indicator-upower #14: Indicator doesn't work with mobile thethering enabled

Caedorus, 2025-01-12:


Danfro, 2025-01-12:
:thinking: how do you enable tethering. I don't see this option on my devices. Or are you still on xenial?

Caedorus, 2025-01-12:
I have UT Tweak Tool installed from the openstore, in there press three dots left-top select System then ADB settings

Caedorus, 2025-01-12:
Some screenshots
[image]
[image]
[image]

Danfro, 2025-01-12:
Yeah, I was looking there. But which of this is tethering? :thinking: I thought that was named like that, with something tethering in it.

Caedorus, 2025-01-12:
RNDIS is mobile tethering, sorry for the confusion

Danfro, 2025-01-12:
Also did you try to reinstall the indicator? Maybe that could help?

Danfro, 2025-01-12:
At least on my suzu it seems to work fine even with tethering enabled. I have the phone plugged into my laptop and also did a reboot.

Caedorus, 2025-01-12:
So, I enable it, reboot and indicator is broken; disable it, reboot again and indicator works again

Caedorus, 2025-01-12:
I'll try reinstalling it

Caedorus, 2025-01-12:
Ok, now it works... I feel stupid and sorry for wasting your time

closed 2025-01-12 by Caedorus

Danfro, 2025-01-12:
No problem. You might not be the only one. But I tested also on my OP5T and it works fine. So might have been just a software hiccup on your end.