paulcarroty/indicator-upower #16: support for the fairphone 4

Fuseteam, 2025-01-14:
Hi i would like to help get upower working on the fairphone 4
running the script i get this
[power.txt](https://github.com/user-attachments/files/18410315/power.txt)

I also had these errors:
```
cat: /sys/devices/virtual/thermal/thermal_zone26/temp: Invalid argument
cat: /sys/devices/virtual/thermal/thermal_zone35/temp: Invalid argument
cat: /sys/devices/virtual/thermal/thermal_zone25/temp: Invalid argument
cat: '/sys/devices/platform/soc/990000.i2c/i2c-2/2-0034/990000.i2c:qcom,smb1396@34:qcom,div2_cp/power_supply/charge_pump_master/cp_die_temp': No data available
cat: '/sys/devices/platform/soc/c440000.qcom,spmi/spmi-0/spmi0-02/c440000.qcom,spmi:qcom,pm7250b@2:qcom,qpnp-smb5/power_supply/battery/charger_temp': No data available
cat: /sys/devices/virtual/thermal/thermal_zone26/temp: Invalid argument
cat: /sys/devices/virtual/thermal/thermal_zone35/temp: Invalid argument
cat: /sys/devices/virtual/thermal/thermal_zone25/temp: Invalid argument
```

closed 2025-01-16 by paulcarroty

Fuseteam, 2025-01-16:
oh cool, how do i test it?

paulcarroty, 2025-01-16:
Test builds are here: https://github.com/paulcarroty/indicator-upower/actions/workflows/ci.yml