paulcarroty/indicator-upower #36: Data fot device Oneplus Nord N100 (On Noble)

ChromiumOS-Guy, 2025-09-21:
[power.txt](https://github.com/user-attachments/files/22452476/power.txt)

ChromiumOS-Guy, 2025-09-24:
while charging from usb (same port/pc/cable):
```
phablet@ubuntu-phablet:~$ find /sys/devices/ -iname "*parallel*" -type f -exec echo "{}" \; -exec cat {} \;
/sys/devices/platform/soc/4a84000.i2c/i2c-0/0-000c/4a84000.i2c:qcom,smb1355@c:qcom,smb1355-charger@1000/power_supply/parallel/parallel_mode
1
/sys/devices/platform/soc/4a84000.i2c/i2c-0/0-000c/4a84000.i2c:qcom,smb1355@c:qcom,smb1355-charger@1000/power_supply/parallel/parallel_fcc_max
6000000
/sys/devices/platform/soc/4a84000.i2c/i2c-0/0-000c/4a84000.i2c:qcom,smb1355@c:qcom,smb1355-charger@1000/power_supply/parallel/parallel_batfet_mode
1
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qcom,qpnp-smb5/power_supply/battery/parallel_current_now
-2594
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qcom,qpnp-smb5/power_supply/battery/parallel_disable
0
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2594
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_power_v_i_parallel_vbat_vdata_raw
22282 65525 
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_voltage_i_parallel_raw
65527
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_voltage_i_parallel_input
-1678
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_power_v_i_parallel_vbat_vdata_input
4330665 -1678 
```
while not connected:
```
phablet@ubuntu-phablet:~$ find /sys/devices/ -iname "*parallel*" -type f -exec echo "{}" \; -exec cat {} \;
/sys/devices/platform/soc/4a84000.i2c/i2c-0/0-000c/4a84000.i2c:qcom,smb1355@c:qcom,smb1355-charger@1000/power_supply/parallel/parallel_mode
1
/sys/devices/platform/soc/4a84000.i2c/i2c-0/0-000c/4a84000.i2c:qcom,smb1355@c:qcom,smb1355-charger@1000/power_supply/parallel/parallel_fcc_max
6000000
/sys/devices/platform/soc/4a84000.i2c/i2c-0/0-000c/4a84000.i2c:qcom,smb1355@c:qcom,smb1355-charger@1000/power_supply/parallel/parallel_batfet_mode
1
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qcom,qpnp-smb5/power_supply/battery/parallel_current_now
-2441
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qcom,qpnp-smb5/power_supply/battery/parallel_disable
0
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_power_v_i_parallel_vbat_vdata_raw
22040 65522 
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_voltage_i_parallel_raw
65520
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_voltage_i_parallel_input
-2441
/sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:vadc@3100/iio:device0/in_power_v_i_parallel_vbat_vdata_input
4293489 -2136 
```

ChromiumOS-Guy, 2025-09-25:
NOT CONNECTED TO ANYTHING
idle
```
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441

```
under load (4k video on yt)
fluctuates widely
```
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2594
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-1831
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-1831
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2136
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-3204
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2594
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2899
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-1831
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-1831
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-1678
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-3204
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2594
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2136
phablet@ubuntu-phablet:~$ 
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2899
phablet@ubuntu-phablet:~$ cat /sys/devices/platform/soc/1c40000.qcom,spmi/spmi-0/spmi0-02/1c40000.qcom,spmi:qcom,pmi632@2:qpnp,qg/power_supply/bms/parallel_current_now
-2441
phablet@ubuntu-phablet:~$ 
```

paulcarroty, 2025-09-25:
`-1831` on load, so lower than `-2136` on idle? Weirddd... Do you have any hardware tester to measure the current on charge?

ChromiumOS-Guy, 2025-09-27:
nope, i mean kind of i have an oscilloscope but im not going to wreck a perfectly good charging cable for this.
i mean the phone charges/discharges/displays charge fine so i assume it works.

paulcarroty, 2025-11-15:
@ChromiumOS-Guy should I revert https://github.com/paulcarroty/indicator-upower/commit/c0a1ff9f126e6a43084f85cb8e4205699469ef9f ?